majbthrd/pico-debug #2: Missing serial and manufacturer information

UweBonnes, 2021-02-11:
Hello,

Blackmagic probe for PC supports CMSIS DAP probe. I tried both uf2s download from Releases. My recognition for CMSIS Dap stumbles on the missing iManufacturer and iSerial entries.  It is a good habit to provide these.

DCelectronics, 2021-02-11:
@UweBonnes For the blackmagic probe (BMP) wouldn't we need to create RP2040 .h and .c files in the [target folder](https://github.com/blacksphere/blackmagic/tree/master/src/target) as well as include them in the [Makefile](https://github.com/blacksphere/blackmagic/blob/master/src/Makefile), and recompile from source to add support for this chip?
Although I have a BMP, I have been successful to run a second RP Pico board as a picoprobe with openocd.

UweBonnes, 2021-02-12:
Implementing multidrop (only bmp and libftdi for now) and adding the target is done at https://github.com/UweBonnes/blackmagic/tree/rp and awaiting user testing, not yet implementing flash handling. Now I want to use  BMP hosted with pico-debug. If sombody likes, also a backend for   picoprobe could be implemented.

majbthrd, 2021-02-13:
@UweBonnes, I'm not aware of any USB requirement that a USB device must have an iManufacturer and iSerial string.  Omitting them is a legitimate USB configuration.  Even a device name isn't a USB requirement, but it is for CMSIS-DAP in order for the debugger to be detected.  pico-debug is designed to fit into as small as possible a RAM image, and adding unnecessary functionality hurts this objective.

Please take a look at the sample code I provided you in [one of the other issues you created]( https://github.com/majbthrd/DapperMime/issues/1).  You can also look at OpenOCD, where it just so happens that I contributed the PR for CMSIS-DAP detection some years ago.

closed 2021-02-14 by majbthrd