majbthrd/pico-debug #3: Question about the printf statements

pradt2, 2021-02-21:
Hi,

Given that the debugger 'owns' the USB port, how can I see the _printf_ statements?

majbthrd, 2021-02-21:
Great question.

One option is to connect a USB TTL serial port to the UART that pico-sdk has been configured to send printf messages out of.

For at least the past 15+ years in professional debugging products, the SWD/JTAG connection can be used for printf statements directly.  This is a much better way of doing things IMHO.  The way it works is that the user code writes to a location in (RP2040 Pico) memory, and then the PC software can read that memory whilst the user code is running.  However, that’s not a feature that open source solutions like OpenOCD, GDB, etc. support yet (to my knowledge).  My hope is that pico-debug might spur these kinds of features to be added; they would benefit anyone doing debugging (not just RP2040 Pico).

majbthrd, 2021-02-21:
P.S.: I should point out that when I say "SWD/JTAG", I mean the debugger connection (whether provided by pico-debug or something else).

majbthrd, 2021-02-21:
Also, it is perhaps worth pointing out that a debugger (like gdb or Visual Studio Code) works with pico-debug to let the user see variable values and set breakpoints, etc.

So, you might find that printf statements are not as relevant when you can leverage more advanced debugger capabilities.  Why printf a particular value when you can see its value (and every other variable’s value) in the debugger running on the PC?  Why have a printf to show when code reaches a particular point when you can set a breakpoint?  Etc., etc., etc.

closed 2021-02-24 by majbthrd

pradt2, 2021-02-24:
Thanks very much for these answers, I think you can't always replace variable inspection with _printf_ statements (e.g. in my case I use printf to see how sensor readouts change in real time), but you're completely right - if I had to pick one, I'd choose a debugger any time of day.

majbthrd, 2021-02-24:
Cool.  FYI: [Rowley Crossworks for ARM](https://www.rowley.co.uk/arm/index.htm) and [SEGGER Embedded Studio](https://www.segger.com/products/development-tools/embedded-studio/) (white-label version of Rowley) have quite powerful functionality for writing data via the debugger link.  The APIs are here:

https://studio.segger.com/debugio_h.htm
https://www.rowleydownload.co.uk/arm/documentation/debugio_h.htm

The notion that a user can fopen and fwrite a file on the host PC from the embedded target is pretty cool to me.

I use Rowley Crossworks for ARM almost daily, and there is example RP2040 code [here](https://github.com/majbthrd/pico-demos).  I haven't yet tried SES with the RP2040, but in theory they do now support CMSIS-DAP/DAP-Link (originally, SEGGER Embedded Studio only worked with J-Link).