majbthrd/pico-debug #5: openocd build for OSX Users

citizenfish, 2021-03-13:
Firstly thanks for all the good work here

Just wanted to add some  extra notes for Mac users  following your openocd build notes

If you get the error:-

```bash
configure: error: hidapi is required for the CMSIS-DAP Compliant Debugger

```

Then take the following steps:-

```bash
git clone https://github.com/signal11/hidapi.git
cd hidapi
./configure --enable-static --disable-shared
make

export HIDAPI_LIBS="-L`pwd`/mac/.libs -lhidapi"
export PKG_CONFIG_PATH="`pwd`/pc"
```

Then change back to the openocd directory and configure as per your instructions. 
The environment variables above will ensure that hidapi is found and linked.

majbthrd, 2021-03-13:
That is much appreciated!  I have revised howto/openocd.md to link to this page.

closed 2021-03-13 by majbthrd

frederikvs, 2021-03-26:
FYI : on my Mac I've also had to change the config command for openocd to

```
./configure --enable-cmsis-dap CFLAGS="-Wno-deprecated -Wno-strict-prototypes"
```

Apparently the compiler on some Macs is more sensitive to certain types of warnings.

ZontarOne, 2021-06-19:
I also had to include hidapi headers before configuring openocd:

```
 export HIDAPI_CFLAGS="-I /`pwd`/hidapi"
```