majbthrd/pico-debug #6: cmsis-dap

vorosj, 2021-03-19:
After dropping in the pico-debug-gimmecache.uf2, it reboots as cnsis-dap, but openocd can't find it:
```
C:\0work\rp2040\openocd_picoprobe>openocd -f cmsis-dap.cfg -c "transport select swd" -c "adapter speed 4000" -f target/rp2040-core0.cfg
Open On-Chip Debugger 0.10.0+dev-g14c0d0d-dirty (2021-01-27-15:43)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
Error: The specified debug interface was not found (cmsis-dap)
The following debug adapters are available:
```
I can't find cmsis-dap in the Device Manager. 
Zadig find it, not sure what is the proper driver (if any).
Could you please provide a bit more detailed (step by step) instruction how to use it in Segger Studio? Your demo program compiled in Studio, but not sure how to debug it (either with pico-debug or jlink).
regards,
   Jozsef

majbthrd, 2021-03-19:
What Operating System are you using?

vorosj, 2021-03-19:
Maybe my openocd is not the correct version. Is there a binary available for win10?

vorosj, 2021-03-19:
Wow, that was fast, thank you. win10 is my operating system.

vorosj, 2021-03-19:
No new HID device in the Win10 Device Manager after Pico reboots as CMSIS_DAP.

majbthrd, 2021-03-19:
> No new HID device in the Win10 Device Manager after Pico reboots as CMSIS_DAP.

I see it (see below); are you sure?

[image]

vorosj, 2021-03-19:
I count 14 devices before dropping pico-debug-gimmecache.uf2, and 14 after. Zadig list it.
[image]
[image]
[image]

vorosj, 2021-03-19:
Zadig found 13 devices without Pico-debug, and 14 with it, but Device Manager lists 14 HID devices with and without.

majbthrd, 2021-03-19:
Zadig is NOT to be used with CMSIS-DAP.  CMSIS-DAP is a vendor-defined HID device.  If you have used Zadig to install a libusb device driver, then Windows will no longer see that device as a vendor-defined HID device.

majbthrd, 2021-03-19:
I think the problems that you are experiencing are due to the following:

1) you decided to use Zadig to install a driver (which you should not do)
2) the OpenOCD Windows binaries that you are using are not based on the source code pointed to in the pico-debug instructions

Sorry, but I do not have pre-built OpenOCD Windows binaries.

closed 2021-03-19 by majbthrd

vorosj, 2021-03-19:
I tried Zadig after I didn't found the cmsis-dap in the Device Manager. Now I know it was a mistake.
How can I uninstall the driver Zadig installed to Pico cmsis-dap?

For test reason, I flashed a BluePill to cmsis-dap, jut to see, how it looks like in Device Manager. It appears as 2 HID device. I am not familiar with cmsis-dap, I use stlink and jlink for embedded development.

I wish either PlatformIO, Keil or Segger Studio provides full support for the RP2040. This openocd route is extremely complicated. Is there a openocd binary available for Pico-Debug? I understand, that compiling tools is common in Linux, but it's not the case in win. I tried VSCode and the Pico toolset on a Pi4, worked, but that thing is just too slow for serious work.

vorosj, 2021-03-19:
Ok, I wrote it before reading your last post, sorry.
So far i have some luck with the PlatforIO Wizio plugin and with your demo in Segger. In both cases compiling is working, but can't debug because there are no support for debug configuration. However, Ozone is working with JLInk, so I can debug using the .elf.

majbthrd, 2021-03-19:
I am not familiar with Zadig; if you do a web search for "remove zadig drivers", there are various things you can try.

I *COMPLETELY* agree that openocd is a pain to use.  

I've been very happy with Rowley Crossworks for ARM, which absolutely works directly with pico-debug.  Furthermore, Crossworks is what SEGGER Embedded Studio was originally based on.

SEGGER Embedded Studio claims to work directly with CMSIS-DAP:

https://www.segger.com/news/segger-embedded-studio-now-supports-dap-link/

However, it doesn't seem to work (which is why I had to write the instructions for using OpenOCD); when I asked SEGGER why they say it works when it doesn't, they don't reply:

https://forum.segger.com/index.php/Thread/7848-What-happened-to-CMSIS-DAP-DAP-Link-support/

majbthrd, 2021-03-19:
I took a look at adding a Keil project.

Because pico-debug is a CMSIS-DAP adapter, Keil directly supports it (no OpenOCD needed):

[image]

**HOWEVER**, Keil does not have a "pack" for RP2040.  The "pack" includes a "flash download" driver, which is needed to download the user application to flash.

It is technically possible to write a new flash download algorithm:

https://www.keil.com/pack/doc/CMSIS/Pack/html/flashAlgorithm.html

but "Creating a Flash programming algorithm with MDK-Lite is not supported.", so that person would not only have to write and debug the RP2040 flash download driver, but they would also have to buy a Keil MDK commercial license for the privilege to do so!!!! :(