majbthrd/pico-debug #7: Add USB serial number string

flit, 2021-03-19:
Not having a serial number string makes it difficult to identify uniquely and support multiple connected probes.

A short manufacturer string would be nice, too.

I know they take up memory, but it's useful.

majbthrd, 2021-03-21:
I have added a unique ID (using the SPI flash 0x4B command) that appears both in the USB descriptor and CMSIS-DAP’s DAP_Info command.  I have also added a manufacturer string.

The size of the code to retrieve the unique ID from the SPI flash dwarfs the size of storing the string that ultimately contains the result.

The way to make things fit within the existing pico-debug footprint was to cheat somewhat.  Since pico-debug loads as an SRAM .uf2, it has to load into SRAM for the Boot ROM to run.  Upon initial execution, it briefly has access to the entire SRAM (and not just the portion of memory earmarked for pico-debug’s use).  The code for retrieving the unique ID (as well as PLL initialization code previously resident in the app that I moved to free up string space) executes first, and once it has written the result into the string buffer used by pico-debug to store the unique ID, it has done its job.  The .uf2 image is slightly larger, but only the existing smaller footprint is needed for debugger operation.

closed 2021-03-21 by majbthrd

flit, 2021-03-21:
Great! Thanks for adding this. And I wouldn't call that cheating—it's a well-crafted hack. 😉