majbthrd/pico-debug #10: openocd build error for macOS

Dracanrage, 2021-05-19:
I have referred to the #5, but it didn't work for me.
After used that commands, it shows the `error: hidapi is required for the CMSIS-DAP Compliant Debugger` as before.
[image]
[image]

majbthrd, 2021-05-19:
Note that the first screenshot shows:

`
checking for hidapi... no
`

which implies the steps prescribed in #5 didn't happen properly.  I have to wonder whether your "export" commands in the second screenshot may not have created the environmental variables.  You can run "export" without arguments to see what HIDAPI_LIBS and PKG_CONFIG_PATH got set to.

@citizenfish, would you be able to provide any insight?  I am not a Mac user.  Thanks.

citizenfish, 2021-05-19:
I have edited my instructions which may be causing this issue.

Run ./bootstrap then ./configure --enable-static --disable-shared

But as @majbthrd said we need to see the output of export after the two commands you should see something like this:-

```
declare -x PKG_CONFIG_PATH="/Users/daveb/dev/scratch/hidapi/pc"
declare -x HIDAPI_LIBS="-L/Users/daveb/dev/scratch/hidapi/mac/.libs -lhidapi"
```
And be able to list the two directories with ls

Dracanrage, 2021-05-19:
> I have edited my instructions which may be causing this issue.
>
> Run ./bootstrap then ./configure --enable-static --disable-shared
>
> But as @majbthrd said we need to see the output of export after the [...]

[image]
I use echo and get this, but I didn't see like `declare ......`

Dracanrage, 2021-05-19:
After I use 
```
declare -x PKG_CONFIG_PATH="/Users/mac/openocd/hidapi/pc"
declare -x HIDAPI_LIBS="-L/Users/mac/openocd/hidapi/mac/.libs -lhidapi"
```
It didn't work as well
[image]
[image]

majbthrd, 2021-05-19:
@Dracanrage, "declare" is not a shell command.  If you scroll back, you will see that @citizenfish stated this what you should see output after the "export" command.

@Dracanrage, you have not showed a screenshot of following the #5 instructions for building hidapi where "./configure --enable-static --disable-shared" is used.

closed 2021-05-19 by majbthrd

Dracanrage, 2021-05-20:
> @Dracanrage, "declare" is not a shell command. If you scroll back, you will see that @citizenfish stated this what you should see output after the "export" command.
>
> @Dracanrage, you have not showed [...]
@majbthrd 
Sorry, I'm in the different time zone, I just read the response before.
1. git clone https://github.com/signal11/hidapi.git
[image]
2. cd hidapi
[image]
3. ./bootstrap
then I get this:
[image]
if I use the hidapi from libusb/hidapi, I get this:
[image]
[image]
[image]
It looks well now.
4. ./configure --enable-static --disable-shared
[image]
[image]
5. make
[image]
6. export HIDAPI_LIBS="-L`pwd`/mac/.libs -lhidapi" and export PKG_CONFIG_PATH="`pwd`/pc"
[image]
I know @citizenfish says will see some thing like this:
```
declare -x PKG_CONFIG_PATH="/Users/daveb/dev/scratch/hidapi/pc"
declare -x HIDAPI_LIBS="-L/Users/daveb/dev/scratch/hidapi/mac/.libs -lhidapi"
```
But I didn't see anything, so I tried them as shell commands to figure out whether it could be used.
7. ./configure --enable-cmsis-dap CFLAGS="-Wno-deprecated -Wno-strict-prototypes"
[image]
[image]
Then it crashed as I said.

citizenfish, 2021-05-20:
It looks to me like you may be this stage wrong by pasting in both commands at once

[image]

Follow these instructions:-

1) enter the command below then press return
```
export HIDAPI_LIBS="-L`pwd`/mac/.libs -lhidapi"
```
2) enter the command below then press return
```
export PKG_CONFIG_PATH="`pwd`/pc"
```
3) enter this command then press return
```
ls $PKG_CONFIG_PATH
```

You should see a directory listing, if not enter this command then press return and show us the result

```
echo $PKG_CONFIG_PATH
```

Dracanrage, 2021-05-20:
I have seen the directory as below:
[image]
But when I use 

```
./configure --enable-cmsis-dap CFLAGS="-Wno-deprecated -Wno-strict-prototypes"
```
It still didn't work:
[image]
[image]

majbthrd, 2021-05-20:
Doing a web search for “macos openocd cmsis-dap hid” yields this:

https://gist.github.com/technobly/f2c14eaa7334db80849d45614250bdf0

The author's instructions call for setting the paths in the same line as part of the configure call, and the author writes "Seems gross to copy the path directly, but trust me it doesn't work unless you do".

Please understand that I am NOT SUGGESTING that you copy and paste the instructions exactly as-is, but perhaps this might suggest a path for you to explore for the problem with your particular Mac.

Dracanrage, 2021-05-23:
> Doing a web search for “macos openocd cmsis-dap hid” yields this:
>
> https://gist.github.com/technobly/f2c14eaa7334db80849d45614250bdf0
>
> The author's instructions call for setting the paths in the [...]

Thanks! I finally find the problem:
The hombrew didn't really install the pkg-config package which openocd required.
Just try:

```
brew reinstall pkg-config
``` 

Then re-configure, all will be fine.