majbthrd/pico-debug #12: pico-debug.cfg? where is it?

hgrodriguez, 2021-08-07:
Hi,
I do not know, how else I could contact you.

I found this software, and at the page:
https://github.com/majbthrd/pico-debug/blob/master/howto/openocd.md
you mention to run openocd like:
openocd -f board/pico-debug.cfg

but:

1. I cannot find the pico-debug.cfg contents anywhere.
2. I tried to start this with:
bin/openocd -f scripts/interface/cmsis-dap.cfg
but I get the error  JTAG not supported:
xPack OpenOCD, x86_64 Open On-Chip Debugger 0.11.0-00155-ge392e485e (2021-03-15-18:44)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
cmsis-dap
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Info : CMSIS-DAP: SWD  Supported
Info : CMSIS-DAP: FW Version = 2.0.0
Info : CMSIS-DAP: Serial# = E6603828236E6935
Error: CMSIS-DAP: JTAG not supported

How to I get rid of this?
As I understand, only SWD is needed, but I could not find out how to remove JTAG.

Thanks in advance,
Holger

PS: I spent hours now in trying to debug RP2040 on macOS, and I am now desperate to get anything going...

majbthrd, 2021-08-07:
@hgrodriguez, you have to follow the entire instructions, not just selective portions of it.

I know that you did not follow the instructions because your sample output shows you are using too old a version of OpenOCD.  It looks like you are using an "xPack OpenOCD" branded version from March 2021, which is not what the instructions indicate.

The instructions that you linked to have two sections "pre-build openocd" and "Build openocd yourself".  You must choose to follow one of these.

You cannot just copy board/pico-debug.cfg to your existing "xPack OpenOCD" and expect it to work.  You need one of the newer OpenOCD versions described in the two sections "pre-build openocd" and "Build openocd yourself".

Chances are that you are trying to debug something that you wrote with Pico SDK, and when you try to debug it, you'll find things don't work, and then when you desperately ask why, I'll have to point out that you also didn't read the pico-debug instructions saying that you can't use USB on your application when you are also using pico-debug.

If you want an easy to use example of using pico-debug, try [arduino-pico](https://github.com/earlephilhower/arduino-pico).

closed 2021-08-07 by majbthrd

marcj, 2022-09-14:
Quick tip for anyone who stumbles across this problem when using macOS:

When installing open-ocd with brew  via `brew install openocd` you get an old version **that is not compatible** with pico-debug, which means it has not the `board/pico-debug.cfg` file  (at least at the moment). To get a new version you can compile manually or use 

```sh
$ brew install --HEAD openocd
```

then you will have the file `board/pico-debug.cfg`.

``` sh
$ brew ls openocd
/opt/homebrew/Cellar/open-ocd/HEAD-219cb95/bin/openocd
/opt/homebrew/Cellar/open-ocd/HEAD-219cb95/share/info/ (3 files)
/opt/homebrew/Cellar/open-ocd/HEAD-219cb95/share/man/man1/openocd.1
/opt/homebrew/Cellar/open-ocd/HEAD-219cb95/share/openocd/ (892 files)

$ ls -al /opt/homebrew/Cellar/open-ocd/HEAD-219cb95/share/openocd/scripts/board/pico-debug.cfg
```

you can then run it like

```sh
$ /opt/homebrew/Cellar/open-ocd/HEAD-219cb95/bin/openocd -f board/pico-debug.cfg
Open On-Chip Debugger 0.11.0+dev-00873-g219cb9598 (2022-09-14-14:05)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
cortex_m reset_config sysresetreq

Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Info : CMSIS-DAP: SWD supported
Info : CMSIS-DAP: Atomic commands supported
Info : CMSIS-DAP: FW Version = 2.0.0
Info : CMSIS-DAP: Serial# = E6616407E3568E25
Info : CMSIS-DAP: Interface Initialised (SWD)
Info : SWCLK/TCK = 0 SWDIO/TMS = 0 TDI = 0 TDO = 0 nTRST = 0 nRESET = 0
Info : CMSIS-DAP: Interface ready
Info : clock speed 4000 kHz
Info : SWD DPIDR 0x0bc12477
Info : [rp2040.cpu] Cortex-M0+ r0p1 processor detected
Info : [rp2040.cpu] target has 4 breakpoints, 2 watchpoints
Info : starting gdb server for rp2040.cpu on 3333
Info : Listening on port 3333 for gdb connections
```