majbthrd/pico-debug #13: Can I include the two pico-debug-xxxx.uf2 in my repository?

GorgonMeducer, 2021-09-18:
Hi,

Thank you for this fantastic job! 
I have made a Pico template for MDK, and it works well!

https://github.com/GorgonMeducer/Pico_Template

I have some questions:

- Can I add the two uf2 files, i.e. pico-debug-XXXX.uf2, to my repository?
- If it is allowed, what should I do for licensing issue? I need your guidance.

Thank you in advance.

Cheers,
Gabriel

majbthrd, 2021-09-18:
Cool!

It seems like the simplest solution is that you just link to the [pico-debug download](https://github.com/majbthrd/pico-debug/releases).

That way, users can always get the most current version.  This is what other projects like [arduino-pico](https://github.com/earlephilhower/arduino-pico) have done.

You mentioned license; the pico-debug license is specified in the README.md of the project.  It uses the MIT and Apache 2.0 licenses.  If you think it is important to copy the uf2 files to your own repository, then please follow those license terms for copies of object code.

closed 2021-09-18 by majbthrd

GorgonMeducer, 2021-09-18:
I will put a link to the pico-debug download.

Thank you.