majbthrd/pico-debug #15: Any way to use pico-debug over UART?

MattMills, 2021-12-08:
I need to debug code that uses tinyUSB for it's core functionality (basically, re-implementing the usb mass storage flash interface in normal userspace code alongside serial interfaces), is there any way to use pico-debug without using the USB interface? 

Or alternatively, using it via a second pico with an interconnect to SWD and SWCLK?

So far pico-debug is the only debugger I've gotten to work, as picoprobe doesn't seem to have any documentation.

MattMills, 2021-12-08:
Actually, I thought of a way to rework the code to test it without tinyUSB and the debugger stops responding as soon I hit my area of interest (a call to flash_range_erase), so while it'd be nice of have a UART based interface it probably won't help me much.

majbthrd, 2021-12-08:
> I need to debug code that uses tinyUSB for it's core functionality (basically, re-implementing the usb mass storage flash interface in normal userspace code alongside serial interfaces), is there [...]

No, there is no way to use pico-debug over a UART.

If you have a second pico, you might want to try [Dapper Mime](https://github.com/majbthrd/DapperMime).  It wires in exactly the same as picoprobe, but uses a standard protocol (CMSIS-DAP, the same as what pico-debug uses) instead of a proprietary one.

closed 2021-12-08 by majbthrd

MattMills, 2021-12-08:
Thanks, I've gotten it working in a few minutes after setting up Dapper Mime. You probably already know, but just FYI, Dapper Mime appears to pull all the TinyUSB dependencies 2 or 3 times which makes it quite slow to init.