majbthrd/pico-debug #17: Crash after a few steps with simple blink

nono313, 2022-03-20:
I am unable to step through a simple blink program on my Raspberry Pi Pico using pico-debug (gimmecache).

Here are the steps I am taking;
- Connect the Pico with BOOT button pressed
- Copy pico-debug-gimmecache.u2f to the mass storage of the Pi Pico
- Start openocd with the following command : `openocd -f board/pico-debug.cfg ` From there, openocd starts correctly (version 0.11.0) on port 3333
- Start gdb with the following command : `arm-none-eabi-gdb blink.elf` 
- Then I enter ` tar ext :3333` in the gdb console
- I run `load`, then ` break main`. Here I get a message with the following note : ` Note: automatically using hardware breakpoints for read-only addresses.`
-  ` continue`

As soon as I enter an `s` command to step through the code, I get the following error message in the gdb console : 

```
CMSIS-DAP command CMD_DISCONNECT failed.
error writing data: No such device
CMSIS-DAP command CMD_CONNECT failed.
error writing data: No such device
CMSIS-DAP command CMD_DISCONNECT failed.
error writing data: No such device
CMSIS-DAP command CMD_CONNECT failed.
error writing data: No such device
error writing data: No such device
```
I see the same messages in the openocd window,

majbthrd, 2022-03-20:
This is a classic 'user program pulls the rug out from under pico-debug'.  Since your program and pico-debug share the same CPU, if your program misbehaves in particular ways, pico-debug is disrupted.  If this is not to your liking, I'd encourage you to purchase a separate debug pod and wire it to your RP2040.

There is a section entitled "Caveats whilst using pico-debug" in the main README.md.  Your "simple blink" app is doing one or more of the items listed there.

To start with a safe know good example, I would encourage you to follow the steps here (using "picoboard_blinky" sample):

https://github.com/majbthrd/pico-debug/blob/master/howto/openocd.md

closed 2022-03-20 by majbthrd

nono313, 2022-03-21:
Thanks for the response.

My original test was with the pico example blinky.

I am now trying with the picoboard blinky : At first it seems to work a bit better, but after a few seconds stepping and going run and pause, I still get the CMD_CONNECT / CMD_DISCONNECT errors.

Is the picoboard_blinky supposed to work flawlessly with pico-debug ?

majbthrd, 2022-03-21:
> I am now trying with the picoboard blinky : At first it seems to work a bit better, but after a few seconds stepping and going run and pause, I still get the CMD_CONNECT / CMD_DISCONNECT errors.
> 
> Is the picoboard_blinky supposed to work flawlessly with pico-debug ?

Yes

nono313, 2022-03-21:
Ok so why is it not working for me ? How can I make it work or get more helpful information on what is happening ?

majbthrd, 2022-03-21:
I'm also running openocd 0.11.0, and I just re-downloaded pico-sdk and pico-examples in case there was some recent effort on those projects to break pico-debug.  In the following example, I did a git clone to create these two directories:

/tmp/pico-sdk
/tmp/pico-examples

$ cd /tmp/pico-examples
$ mkdir build
$ cd build
$ cmake -DCMAKE_BUILD_TYPE=Debug -DPICO_SDK_PATH=/tmp/pico-sdk ..
$ make picoboard_blinky
$ cd picoboard/blinky

At this point, I start the RP2040 with the pushbutton pressed, and once Linux *finally* mounts it, I copy pico-debug:

$ cp pico-debug-gimmecache.uf2 /media/phl/RPI-RP2/

and then run openocd:

$ openocd -f board/pico-debug.cfg

which gives something like this:

```
Open On-Chip Debugger 0.11.0+dev-00347-g3f1c15d2a-dirty (2021-08-24-09:20)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
cortex_m reset_config sysresetreq

Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Info : CMSIS-DAP: SWD  supported
Info : CMSIS-DAP: Atomic commands supported
Info : CMSIS-DAP: FW Version = 2.0.0
Info : CMSIS-DAP: Serial# = E66038B71372AE39
Info : CMSIS-DAP: Interface Initialised (SWD)
Info : SWCLK/TCK = 0 SWDIO/TMS = 0 TDI = 0 TDO = 0 nTRST = 0 nRESET = 0
Info : CMSIS-DAP: Interface ready
Info : clock speed 4000 kHz
Info : SWD DPIDR 0x0bc12477
Info : rp2040.cpu: hardware has 4 breakpoints, 2 watchpoints
Info : starting gdb server for rp2040.cpu on 3333
Info : Listening on port 3333 for gdb connections
```
In a separate terminal window, I run gdb-multiarch with the newly created .elf:

$ gdb-multiarch picoboard_blinky.elf

(gdb) target remote localhost:3333
(gdb) load
(gdb) monitor reset init
(gdb) b main
(gdb) continue

From then on, use your preferred gdb procedure, ala:

(gdb) s
(gdb) bt
(gdb) continue
Ctrl-C
(gdb) s

etc, etc.

majbthrd, 2022-03-21:
P.S.: the output of cmake looks like this locally:

$ cmake -DCMAKE_BUILD_TYPE=Debug -DPICO_SDK_PATH=/tmp/pico-sdk ..

```
PICO_SDK_PATH is /tmp/pico-sdk
PICO platform is rp2040.
Using regular optimized debug build (set PICO_DEOPTIMIZED_DEBUG=1 to de-optimize)
PICO target board is pico.
Using board configuration from /tmp/pico-sdk/src/boards/include/boards/pico.h
CMake Warning at /tmp/pico-sdk/src/rp2_common/tinyusb/CMakeLists.txt:10 (message):
  TinyUSB submodule has not been initialized; USB support will be unavailable

  hint: try 'git submodule update --init' from your SDK directory
  (/tmp/pico-sdk).

CMake Warning at hello_world/usb/CMakeLists.txt:19 (message):
  not building hello_usb because TinyUSB submodule is not initialized in the
  SDK

Skipping TinyUSB device examples, as TinyUSB unavailable
Skipping TinyUSB host examples, as TinyUSB unavailable
-- Configuring done
-- Generating done
-- Build files have been written to: /tmp/pico-examples/build
```