majbthrd/pico-debug #18: pico-debug is not recognised in Win11 for Arm in Parallel Desktop (for Mac M1)

GorgonMeducer, 2022-03-20:
I use Apple Mac M1 with Parallel Desktop + Win11 for Arm. The cmsis-dap from pico-debug cannot be recognised correctly as shown below:

[image]

meanwhile, the RP1-PR2 can be recognised:

[image]

Pico-debug used to work in Parallel Desktop + Win10 for Arm but I don't know why it doesn't work in Win11.

By the way, other CMSIS-DAP devices can be recognised, so there must be some differences in implementation. 

I know this configuration is very special, but it is the only way I can connect a debugger in Apple Mac M1.

Thank you in advance.

majbthrd, 2022-03-21:
I find it hard to believe that anyone would willingly volunteer to use Windows 11, so it would not surprise me if you happen to be the first pico-debug user with Windows 11.

Even if turns out that this has nothing to do with "Apple Mac M1 with Parallel Desktop" and is just a Windows 11 issue, I don't have a PC that meets the Windows 11 requirements to try things locally.  I also certainly do not have, nor plan to buy, an Apple Mac M1 and Parallel Desktop software.

Let's suppose this is due to Windows 11 being difficult about some aspect of USB implementation that earlier versions were OK with.  The only thing I can come up with is the USB version, which we can change from 2.00 to 1.10 since this is a full-speed device.

If you wanted to help diagnose the issue, here are a few things to try.

1) Try the latest v10.04 release that I've just uploaded today.  This changes the USB version to 1.10; it also adopts the latest tagged TinyUSB implementation as well as updates the bcdDevice to make the change apparent.  Perhaps we will luck out on this being the solution.

2) In your first screen shot, you show a HID Input Device with a warning icon.  Right-click that, select Properties, and look information for what the warning message is.  Being Windows, it will probably be something unhelpful like "Device didn't start", but perhaps this might provide a clue.

3) Download and run USBTreeView (https://www.uwe-sieber.de/usbtreeview_e.html).  Use this to capture information on one or more CMSIS-DAP adapters that do work in your M1+Win11 configuration, and post those in this ticket.  Comparing those to pico-debug might yield some clues.

Thanks.

GorgonMeducer, 2022-03-21:
I didn't know win11 is so notorious, if I did, I would never upgrade to win11. In fact, my windows upgraded to win11 automatically without my permission (I didn't remember I had given such permission).
Let me try those suggestions from you and see what can we do with this problem.

GorgonMeducer, 2022-03-21:
Update:
The latest release works in win11. YES, you are right, `we luck out on this being the solution`.

closed 2022-03-21 by GorgonMeducer