majbthrd/pico-debug #20: Can't Find Any CMSIS-DAP Device

jinyi-x, 2022-04-10:
After I copy uf2 into Pico，it just rebooted and runing previous program, dmesg shows no device has been plugged in after mass storage device disconnected, like this:
'''
 sd 6:0:0:0: [sdb] Attached SCSI removable disk
[  596.680640] usb 1-3: USB disconnect, device number 5
[  615.822719] usb 1-3: new full-speed USB device number 6 using xhci_hcd
[  615.972150] usb 1-3: New USB device found, idVendor=2e8a, idProduct=0003, bcdDevice= 1.00
[  615.972163] usb 1-3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  615.972169] usb 1-3: Product: RP2 Boot
[  615.972173] usb 1-3: Manufacturer: Raspberry Pi
[  615.972176] usb 1-3: SerialNumber: E0C912952D54
[  615.976957] usb-storage 1-3:1.0: USB Mass Storage device detected
[  615.977542] scsi host6: usb-storage 1-3:1.0
[  616.991715] scsi 6:0:0:0: Direct-Access     RPI      RP2              2    PQ: 0 ANSI: 2
[  616.992297] sd 6:0:0:0: Attached scsi generic sg2 type 0
[  616.992998] sd 6:0:0:0: [sdb] 262144 512-byte logical blocks: (134 MB/128 MiB)
[  616.993381] sd 6:0:0:0: [sdb] Write Protect is off
[  616.993386] sd 6:0:0:0: [sdb] Mode Sense: 03 00 00 00
[  616.993748] sd 6:0:0:0: [sdb] No Caching mode page found
[  616.993755] sd 6:0:0:0: [sdb] Assuming drive cache: write through
'''
And, lsusb -t seems doesn't list any new device too.When I try run openocd, it output like this:
'''
Open On-Chip Debugger 0.11.0+dev-00635-g7c6d379cf (2022-04-10-16:45)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
Error: Debug adapter doesn't support any transports?
/usr/local/bin/../share/openocd/scripts/target/rp2040-core0.cfg:3: Error: 
in procedure 'script' 
at file "embedded:startup.tcl", line 26
at file "/usr/local/bin/../share/openocd/scripts/target/rp2040-core0.cfg", line 3
'''

majbthrd, 2022-04-10:
That is not expected behavior.

There is no point in running OpenOCD if lsusb does not show a CMSIS-DAP device is connected.

Please confirm that when you say "I copy uf2 into Pico", is your finger still pressing on the button or do you disconnect power after the copy?

The correct procedure is to disconnect power to the Pico, hold the Pico button, connect power, *stop pressing* the Pico button, wait for your OS to mount the virtual Pico drive, copy over the pico-debug .uf2, and then wait for your OS to mount the CMSIS-DAP device.

After this, lsusb should show a 1209:2488 device.

My dmesg output for that entire procedure looks like this:

```
[ 2252.185695] usb 3-1: new full-speed USB device number 2 using xhci_hcd
[ 2252.354100] usb 3-1: New USB device found, idVendor=2e8a, idProduct=0003, bcdDevice= 1.00
[ 2252.354111] usb 3-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 2252.354115] usb 3-1: Product: RP2 Boot
[ 2252.354117] usb 3-1: Manufacturer: Raspberry Pi
[ 2252.354120] usb 3-1: SerialNumber: E0C912D24340
[ 2252.391991] usb-storage 3-1:1.0: USB Mass Storage device detected
[ 2252.392189] scsi host6: usb-storage 3-1:1.0
[ 2252.392299] usbcore: registered new interface driver usb-storage
[ 2252.394997] usbcore: registered new interface driver uas
[ 2253.400812] scsi 6:0:0:0: Direct-Access     RPI      RP2              1    PQ: 0 ANSI: 2
[ 2253.401488] sd 6:0:0:0: Attached scsi generic sg1 type 0
[ 2253.401936] sd 6:0:0:0: [sdb] 262144 512-byte logical blocks: (134 MB/128 MiB)
[ 2253.404204] sd 6:0:0:0: [sdb] Write Protect is off
[ 2253.404219] sd 6:0:0:0: [sdb] Mode Sense: 03 00 00 00
[ 2253.407387] sd 6:0:0:0: [sdb] No Caching mode page found
[ 2253.407394] sd 6:0:0:0: [sdb] Assuming drive cache: write through
[ 2253.456266]  sdb: sdb1
[ 2253.487273] sd 6:0:0:0: [sdb] Attached SCSI removable disk
[ 2279.790098] usb 3-1: USB disconnect, device number 2
[ 2279.809587] FAT-fs (sdb1): unable to read boot sector to mark fs as dirty
[ 2280.137676] usb 3-1: new full-speed USB device number 3 using xhci_hcd
[ 2280.306407] usb 3-1: New USB device found, idVendor=1209, idProduct=2488, bcdDevice=10.04
[ 2280.306424] usb 3-1: New USB device strings: Mfr=2, Product=1, SerialNumber=3
[ 2280.306429] usb 3-1: Product: CMSIS-DAP
[ 2280.306434] usb 3-1: Manufacturer: pico-debug
[ 2280.306438] usb 3-1: SerialNumber: E66038B71372AE39
[ 2280.343685] hid: raw HID events driver (C) Jiri Kosina
[ 2280.354613] usbcore: registered new interface driver usbhid
[ 2280.354620] usbhid: USB HID core driver
[ 2280.369529] hid-generic 0003:1209:2488.0001: hiddev0,hidraw0: USB HID v1.11 Device [pico-debug CMSIS-DAP] on usb-0000:0e:00.0-1/input0
```

jinyi-x, 2022-04-10:
> That is not expected behavior.
>
> There is no point in running OpenOCD if lsusb does not show a CMSIS-DAP device is connected.
>
> Please confirm that when you say "I copy uf2 into Pico", is your finger [...]

I just download this debugger into my pico, and it does rebooted.Howerver, there is no new usb device shows up. And dmesg just doesn't report any message about usb hid device plugged in.

majbthrd, 2022-04-10:
Is the BOOTSEL button pressed?  If the BOOTSEL button is pressed (or shorted) when the RP2040 reboots, the RP2040 will always run its boot ROM code (which is what your dmesg log shows).

Does you Pico look like this, or is it something else?:

https://www.raspberrypi.com/products/raspberry-pi-pico/

jinyi-x, 2022-04-11:
> Is the BOOTSEL button pressed? If the BOOTSEL button is pressed (or shorted) when the RP2040 reboots, the RP2040 will always run its boot ROM code (which is what your dmesg log shows).
> 
> Does you Pico look like this, or is it something else?:
> 
> https://www.raspberrypi.com/products/raspberry-pi-pico/

Yep, my pico is official board.And I release bootsel when it boot into dfu mode.

jinyi-x, 2022-04-11:
Okay, I managed to solved it. I just forget install rules for openocd.

closed 2022-04-11 by jinyi-x