majbthrd/pico-debug #21: Core 0 with pico-debug resident wakes up at different clock frequency

cbiffle, 2022-05-13:
I expect that pico-debug will need certain resources on the chip to function (the readme lists `PLL_USB`, for instance), but I was surprised to discover that it seems to change the clock config in ways that affects core 0's boot state.

Using an image that doesn't make any changes to clock configuration,
- Booting it from Flash without pico-debug resident sees a core 0 clock frequency of ~6 MHz
- Loading pico-debug (gimmecache, specifically) and issuing `reset` through openocd launches into the same firmware with an apparent clock frequency of... much higher than that.

I assume the debugger switches `CLK_SYS_CTRL` to generate `CLK_SYS` from `PLL_USB`. This would be a good thing to document since it changes the startup behavior of programs.

I'm guessing that most folks aren't noticing this because they're running SDK programs that immediately switch `CLK_SYS` to `PLL_SYS`?

majbthrd, 2022-05-13:
The PLL configuration is shown here:

https://github.com/majbthrd/pico-debug/blob/master/bootcode.c

If you have another clock configuration in mind that still provides the PLL_USB 48MHz clock to allow USB functionality, it is open-source and you could produce your own custom binary to suit your particular needs.

This is the real-world tradeoff of having one core debug the second: given the RP2040 design, both cores share the same peripherals, clock generation, and memory.

If you can afford an external debug pod, that option might be preferable to you.  I've already provided a [RP2040 based option](https://github.com/majbthrd/DapperMime) that can be further customized to run on a variety of TinyUSB-supported microcontrollers.  There are also commercial offerings like the SEGGER J-Link.  The objective of this project is to provide something helpful (within the real-world implementation constraints) to those who might not have such resources available.

cbiffle, 2022-05-13:
Thanks! I have several SWD programmers available to me, but I think your project is incredibly cool and wanted to try it. :-)

Because of how the clock muxes are structured on the RP2040 I suspect that having core 0 wake up at a different speed is effectively unavoidable, as you implied. In case I was unclear, I have no objections to this -- I just think it's worth adding to the documentation to give people a heads up, like the note about the debugger using `PLL_USB`.

majbthrd, 2022-05-14:
I think that I've addressed your concern with the most recent commit.

closed 2022-05-14 by majbthrd

cbiffle, 2022-05-14:
Indeed you have! I appreciate it.

cbiffle, 2022-05-14:
...also, in case anyone reads this bug report in the future, note that I had the cores backwards. pico-debug stays resident on core 0 and launches user code on core 1, not the other way around.