majbthrd/pico-debug #22: OpenBSD compatibility

josuah, 2022-05-21:
OpenBSD has libusb, has OpenOCD build from `master` without significant change (just `sed -ri 's/([^_])(hid_init)/\1hidapi_\2/' src/jtag/drivers/*.c`), and openocd works fine with things like STM32. Great!

With pico-debug attached, I see 
[lsusb.openbsd.txt](https://github.com/majbthrd/pico-debug/files/8747550/lsusb.openbsd.txt).

Comparing with [lsusb.wsl.txt](https://github.com/majbthrd/pico-debug/files/8747551/lsusb.wsl.txt) from #16 (under WSL environment):

```diff
--- lsusb.openbsd.txt	Sat May 21 13:44:27 2022
+++ lsusb.wsl.txt	Sat May 21 13:44:03 2022
@@ -1,29 +1,28 @@
-Bus 000 Device 006: ID 1209:2488 InterBiometrics
+Bus 001 Device 006: ID 1209:2488 Generic Peter Lawrence CMSIS-DAP Dapper Miser
 Device Descriptor:
   bLength                18
   bDescriptorType         1
-  bcdUSB               1.10
-  bDeviceClass            0 (Defined at Interface level)
+  bcdUSB               2.00
+  bDeviceClass            0
   bDeviceSubClass         0
   bDeviceProtocol         0
   bMaxPacketSize0        64
-  idVendor           0x1209 InterBiometrics
-  idProduct          0x2488
-  bcdDevice           10.04
+  idVendor           0x1209 Generic
+  idProduct          0x2488 Peter Lawrence CMSIS-DAP Dapper Miser
+  bcdDevice           10.03
   iManufacturer           2 pico-debug
   iProduct                1 CMSIS-DAP
-  iSerial                 3 E660D4A0A72B4329
+  iSerial                 3 E66038B713297738
   bNumConfigurations      1
   Configuration Descriptor:
     bLength                 9
     bDescriptorType         2
-    wTotalLength           41
+    wTotalLength       0x0029
     bNumInterfaces          1
     bConfigurationValue     1
     iConfiguration          0
-    bmAttributes         0xa0
+    bmAttributes         0x80
       (Bus Powered)
-      Remote Wakeup
     MaxPower              100mA
     Interface Descriptor:
       bLength                 9
@@ -32,8 +31,8 @@
       bAlternateSetting       0
       bNumEndpoints           2
       bInterfaceClass         3 Human Interface Device
-      bInterfaceSubClass      1 Boot Interface Subclass
-      bInterfaceProtocol      1 Keyboard
+      bInterfaceSubClass      0
+      bInterfaceProtocol      0
       iInterface              0
         HID Device Descriptor:
           bLength                 9
@@ -42,33 +41,9 @@
           bCountryCode            0 Not supported
           bNumDescriptors         1
           bDescriptorType        34 Report
-          wDescriptorLength      34
-          Report Descriptor: (length is 34)
-            Item(Global): Usage Page, data= [ 0x00 0xff ] 65280
-                            (null)
-            Item(Local ): Usage, data= [ 0x01 ] 1
-                            (null)
-            Item(Main  ): Collection, data= [ 0x01 ] 1
-                            Application
-            Item(Local ): Usage, data= [ 0x02 ] 2
-                            (null)
-            Item(Global): Logical Minimum, data= [ 0x00 ] 0
-            Item(Global): Logical Maximum, data= [ 0xff 0x00 ] 255
-            Item(Global): Report Size, data= [ 0x08 ] 8
-            Item(Global): Report Count, data= [ 0x40 ] 64
-            Item(Main  ): Input, data= [ 0x02 ] 2
-                            Data Variable Absolute No_Wrap Linear
-                            Preferred_State No_Null_Position Non_Volatile Bitfield
-            Item(Local ): Usage, data= [ 0x03 ] 3
-                            (null)
-            Item(Global): Logical Minimum, data= [ 0x00 ] 0
-            Item(Global): Logical Maximum, data= [ 0xff 0x00 ] 255
-            Item(Global): Report Size, data= [ 0x08 ] 8
-            Item(Global): Report Count, data= [ 0x40 ] 64
-            Item(Main  ): Output, data= [ 0x02 ] 2
-                            Data Variable Absolute No_Wrap Linear
-                            Preferred_State No_Null_Position Non_Volatile Bitfield
-            Item(Main  ): End Collection, data=none
+          wDescriptorLength      32
+         Report Descriptors:
+           ** UNAVAILABLE **
       Endpoint Descriptor:
         bLength                 7
         bDescriptorType         5
@@ -89,3 +64,7 @@
           Usage Type               Data
         wMaxPacketSize     0x0040  1x 64 bytes
         bInterval               1
+can't get device qualifier: Resource temporarily unavailable
+can't get debug descriptor: Resource temporarily unavailable
+Device Status:     0x0000
+  (Bus Powered)
```

So far so good, but I have the same error messae as #16 when running OpenOCD (`Open On-Chip Debugger 0.11.0+dev-g0a36acb-dirty (2022-05-21-13:58)`, I bet that is recent enough :D).

```
$ doas openocd -f board/pico-debug.cfg -c 'debug_level 4'
Open On-Chip Debugger 0.11.0+dev-g0a36acb-dirty (2022-05-21-13:58)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
Warn : Could not determine executable path, using configured BINDIR.
cortex_m reset_config sysresetreq

User : 6 4 options.c:63 configuration_output_handler(): debug_level: 4
User : 7 4 options.c:63 configuration_output_handler():
Info : 8 4 server.c:309 add_service(): Listening on port 6666 for tcl connections
Info : 9 4 server.c:309 add_service(): Listening on port 4444 for telnet connections
Debug: 10 4 command.c:166 script_debug(): command - init
Debug: 11 4 command.c:166 script_debug(): command - target init
Debug: 12 4 command.c:166 script_debug(): command - target names
Debug: 13 4 command.c:166 script_debug(): command - rp2040.cpu cget -event gdb-flash-erase-start
Debug: 14 4 command.c:166 script_debug(): command - rp2040.cpu configure -event gdb-flash-erase-start reset init
Debug: 15 4 command.c:166 script_debug(): command - rp2040.cpu cget -event gdb-flash-write-end
Debug: 16 4 command.c:166 script_debug(): command - rp2040.cpu configure -event gdb-flash-write-end reset halt
Debug: 17 4 command.c:166 script_debug(): command - rp2040.cpu cget -event gdb-attach
Debug: 18 4 command.c:166 script_debug(): command - rp2040.cpu configure -event gdb-attach halt 1000
Debug: 19 4 target.c:1662 handle_target_init_command(): Initializing targets...
Debug: 20 4 semihosting_common.c:130 semihosting_common_init():
Debug: 21 5 cmsis_dap_usb_bulk.c:182 cmsis_dap_usb_open(): enumerating interfaces of 0x8086:0x0000
Debug: 22 7 cmsis_dap_usb_bulk.c:182 cmsis_dap_usb_open(): enumerating interfaces of 0x4b50:0xef9c
Debug: 23 8 cmsis_dap_usb_bulk.c:182 cmsis_dap_usb_open(): enumerating interfaces of 0x8087:0x0a2b
Debug: 24 23 cmsis_dap_usb_bulk.c:182 cmsis_dap_usb_open(): enumerating interfaces of 0x04f2:0xb5ab
Debug: 25 37 cmsis_dap_usb_bulk.c:182 cmsis_dap_usb_open(): enumerating interfaces of 0x0bda:0x0316
Debug: 26 41 cmsis_dap_usb_bulk.c:162 cmsis_dap_usb_open(): found product string of 0x1209:0x2488 'CMSIS-DAP'
Debug: 27 41 cmsis_dap_usb_bulk.c:182 cmsis_dap_usb_open(): enumerating interfaces of 0x1209:0x2488
Debug: 28 41 cmsis_dap_usb_bulk.c:249 cmsis_dap_usb_open(): skipping interface 0, endpoint[0] is not bulk out
Error: 29 68 cmsis_dap_usb_hid.c:209 cmsis_dap_hid_write(): error writing data: (null)
Error: 30 68 cmsis_dap.c:461 cmsis_dap_cmd_dap_info(): CMSIS-DAP command CMD_INFO failed.
Debug: 31 68 command.c:556 run_command(): Command 'init' failed with error code -107
User : 32 68 command.c:619 command_run_line():
Debug: 33 68 target.c:2205 target_free_all_working_areas_restore(): freeing all working areas
```

For now, I have no question or request, just documenting this and gathering clues.
In case that is a bothering, feel free to close this and I will document it somewhere else.

josuah, 2022-05-21:
One question for anyone running a working system: would it be possible to have an `lsusb` output?
It could be useful for anyone searching for `lsusb` through github search box.

josuah, 2022-05-21:
Some extra debug fuel:

```
$ dmesg
...
ugen1 detached
uhidev3 at uhub0 port 1 configuration 1 interface 0 "pico-debug CMSIS-DAP" rev 1.10/10.04 addr 6
uhidev3: iclass 3/1
uhid2 at uhidev3: input=64, output=64, feature=0
```

The `ugen1 detached` is the FAT interface of the bootloader for uploading UF2 getting disconnected.

So the driver used seems to be https://man.openbsd.org/uhid

Here is how the kernel sees the device (without using libusb, OpenBSD's own tool):

```
$ usbdevs -v
Controller /dev/usb0:
...
addr 06: 1209:2488 pico-debug, CMSIS-DAP
         full speed, power 100 mA, config 1, rev 10.04, iSerial E660D4A0A72B4329
         driver: uhidev3
```

josuah, 2022-05-21:
Since the device is not using bulk mode but HID, we can ignore the errors from `cmsis_dap_usb_bulk.c` since [openocd tries all backends in turn](https://github.com/openocd-org/openocd/blob/66335683fec62ac89da48d64932fd9d082314225/src/jtag/drivers/cmsis_dap.c#L53-L61).

This is where the error was generated: https://github.com/openocd-org/openocd/blob/master/src/jtag/drivers/cmsis_dap_usb_hid.c#L206-L211

```C
	/* write data to device */
	int retval = hid_write(dap->bdata->dev_handle, dap->packet_buffer, dap->packet_buffer_size);
	if (retval == -1) {
		LOG_ERROR("error writing data: %ls", hid_error(dap->bdata->dev_handle));
		return ERROR_FAIL;
	}
```

So since [`hid_write`](https://github.com/libusb/hidapi/blob/cd95af8/libusb/hid.c#L1130-L1184) fails without setting an error message, the bug might have to be chased down through libusb down to the [OpenBSD backend](https://github.com/libusb/libusb/blob/ba698478afc3d3a72644eef9fc4cd24ce8383a4c/libusb/os/openbsd_usb.c).

So not an pico-debug issue at first sight.

josuah, 2022-05-21:
Some light in the abyss: I remember that there was some package README for `libusb1`, and here it is:

```
+-------------------------------------------------------------------------------
| Running libhidapi on OpenBSD
+-------------------------------------------------------------------------------

libusb Workaround
=================

*** WARNING ***
libhidapi uses devel/libusb1 to communicate with USB HID devices. uhid(4) and
uhidev(4) need to be disabled in the kernel (see config(8)) or the device will
not be available to libusb. Note that this also deactivates other USB HIDs like
USB keyboard and mouse.
```

This is what I tried: at the boot promtp:

```
>>> boot -c
...
>>> disable uhid
>>> disable uhidev
>>> exit
...
```

And then no external keyboard and no mouse support at all (but internal laptop keyboard ok), and I could get much more interaction with pico-debug on RP2040!

Full verbose log:
[openbsd-no-uhid-no-uhidev-rp2040-pico-debug.txt](https://github.com/majbthrd/pico-debug/files/8747895/openbsd-no-uhid-no-uhidev-rp2040-pico-debug.txt)

As well as a openocd listening on 127.0.0.1:3333 for GDB connection:

```
Proto   Recv-Q Send-Q  Local Address          Foreign Address        TCP-State
tcp          0      0  127.0.0.1.3333         *.*                    LISTEN
tcp          0      0  127.0.0.1.4444         *.*                    LISTEN
tcp          0      0  127.0.0.1.6666         *.*                    LISTEN
...
```

So far so good!

josuah, 2022-05-21:
The only thing that pico-debug could do about this, I suppose, would be using CMSIS-DAPv2 (USB based) instead of v1 (USB HID based).

I might be too lazy to write that patch soon. ^_^'
Maybe it is also something to be fixed on the OpenBSD kernel.

In the meantime, are you interested in me documenting the above in one of the `howto`s of the repo?

majbthrd, 2022-05-21:
I will not be adding CMSIS-DAPv2 to pico-debug...period.  CMSIS-DAPv2 is just bloat, and the whole point of pico-debug is a minimal footprint implementation (so small that it can fit in the cache memory) that co-exists with user programs on the RP2040.

The raison d'être of CMSIS-DAPv2 is to provide a faster transfer mechanism on USB 2.0 High Speed devices (at the cost of ceasing to be a driver-less implementation across all OSs).  The RP2040 is not a USB 2.0 High Speed device, and would not benefit from it.

(Aside: I *might* someday add CMSIS-DAPv2 to DapperMime, since that is a generic CMSIS-DAP implementation on top of TinyUSB (which does support some USB 2.0 High Speed microcontrollers that could benefit from it), but I have more important things to do in life at the moment, and it is not as if there is a shortage of generic CMSIS-DAP / CMSIS-DAPv2 implementations in the world already.)

Despite this being posted as if it were an issue with pico-debug, everything you've written seems to indicate that it is a NOT an issue with pico-debug.  It seems to be an issue of discovering and following the libhidapi README when running OpenOCD on OpenBSD, and judging by web searches for those terms, it looks like you may be the first to document this usage.

Why not consider making a succinct "CMSIS-DAP adapter with OpenOCD on OpenBSD" post under the Discussions section?   I can add a link in howto to that specific post, and perhaps search engines will eventually index it to benefit other CMSIS-DAP OpenBSD users in the future.

closed 2022-05-21 by majbthrd

josuah, 2022-05-21:
> a minimal footprint implementation (so small that it can fit in the cache memory) that co-exists with user programs on the RP2040

And this never ceases to surprise me!

> CMSIS-DAPv2 is just bloat [...] ceasing to be a driver-less implementation across all OSs

This sounds bad. I am new to DAP (and even USB), and did not know which of v1 and v2 was more bloated or simple.

> Despite this being posted as if it were an issue with pico-debug

This being an "Issue" is misleading, I never saw that "Discussion" section before. I certainly did not expect any development effort from you! Sorry if [it](https://github.com/majbthrd/pico-debug/issues/22#issuecomment-1133659928) felt like a request.

> that it is NOT an issue with pico-debug

And I second that it is not!

Not stealing more of your time,
thank you for your answers.

josuah, 2022-08-23:
As discussed on
https://marc.info/?l=openbsd-tech&m=166123550232089&w=2
the Dapper Miser PID and VID codes are being added to the OpenBSD kernel, so that the workaround listed on this thread are not needed anymore.

**Q: would you prefer "pico-debug" or "Dapper Miser" for the PID 0x2488?**

Thank you!

majbthrd, 2022-08-23:
Thanks for pursuing this.

I'm not particularly bothered either way.  It would be more consistent with https://pid.codes/pids/ if it were Dapper Miser.  That's also how it already appears in Linux's lsusb.  I saw no reason to ask for another distinct VID:PID for pico-debug, as it is just yet another CMSIS-DAP implementation.

Note that all CMSIS-DAP implementations **must** have the sub-string "CMSIS-DAP" in the USB Product String.  Ideally, this would be a universal way to detect all CMSIS-DAP implementations in OpenBSD.