majbthrd/pico-debug #24: Error: unable to find a matching CMSIS-DAP device

znmeb, 2022-06-26:
Host is reTerminal (Raspberry Pi Compute Module 4 running 32-bit OS). 

```
❯ pwd
/home/pi/Projects/pico/pico-examples/build/picoboard/blinky
❯ openocd -f ./pico-debug.cfg
Open On-Chip Debugger 0.11.0-g228ede4-dirty (2022-06-24-18:50)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
Info : auto-selecting first available session transport "swd". To override use 'transport select <transport>'.
Info : Hardware thread awareness created
Info : RP2040 Flash Bank Command
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Error: unable to find a matching CMSIS-DAP device
```

`lsusb` shows 

```
❯ lsusb
Bus 001 Device 004: ID 1209:2488 Generic Peter Lawrence CMSIS-DAP Dapper Miser
Bus 001 Device 002: ID 0424:2514 Standard Microsystems Corp. USB 2.0 Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

I copied `pico-debug.cfg` from the `openocd` source tree:

```
❯ cat pico-debug.cfg
# SPDX-License-Identifier: GPL-2.0-or-later
# pico-debug is a virtual CMSIS-DAP debug adapter
# it runs on the very same RP2040 target being debugged without additional hardware
# https://github.com/majbthrd/pico-debug

source [find interface/cmsis-dap.cfg]
adapter speed 4000

set CHIPNAME rp2040
source [find target/rp2040-core0.cfg]
```

What am I missing?

majbthrd, 2022-06-26:
If you add a "-d" to the openocd arguments:

openocd -d -f board/pico-debug.cfg

there will be additional openocd debugging information that might provide some clues as to what is awry.  I would suspect that it has nothing to do with pico-debug per se and would happen with any CMSIS-DAP adapter.

One possibility is that you haven't yet given permission to access said USB device.  I'm assuming that you are familiar with adding udev rules, and here's a udev rule for any and all CMSIS-DAP adapters (including pico-debug).  You'll have to rename it from 98-CMSIS-DAP.rules.txt to 98-CMSIS-DAP.rules (since github wouldn't let me upload it as a .rules):

[98-CMSIS-DAP.rules.txt](https://github.com/majbthrd/pico-debug/files/8987729/98-CMSIS-DAP.rules.txt)

znmeb, 2022-06-27:
Thanks! I added the `udev` rules and tried with `-d`. There was a permissions issue so I ran it again as `root`. Logs attached:

[not-root.log.txt](https://github.com/majbthrd/pico-debug/files/8987753/not-root.log.txt)
[root.log.txt](https://github.com/majbthrd/pico-debug/files/8987754/root.log.txt)

I don't have a hardware debug setup wired but it's looking like I'll need to do so if I want to do any debugging. Troubleshooting `openocd` is a tad above my pay grade when the hardware cost is in the $25US range. :-)

majbthrd, 2022-06-27:
It looks to be similar, if not the same, as another user's prior experience (OS has libhidapi missing or not implemented).  Take a look at #22 and see what he did to solve his problem.

closed 2022-06-27 by majbthrd

znmeb, 2022-06-27:
Thanks! I'm getting a second reTerminal this week so I'll try hacking the kernel on one of them to see if I can save the money I would have spent on a soldering iron.