majbthrd/pico-debug #26: DAP init failed with OpenOCD and arduino-pico

ar1a, 2022-11-13:
Hiya! Trying to get a basic blink working alongside pico-debug. I'm on Windows 11, and am running into this

```
openocd -d2 -s C:\Users\aria\.platformio\packages\tool-openocd-rp2040-earlephilhower/share/openocd/scripts -f interface/cmsis-dap.cfg -f target/rp2040.cfg -c "adapter speed 1000" -c "program {.pio\build\pico\firmware.elf}  verify reset; shutdown;"
Open On-Chip Debugger 0.11.0-g610f137 (2022-02-11-13:57)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
debug_level: 2

Info : auto-selecting first available session transport "swd". To override use 'transport select <transport>'.
Info : Hardware thread awareness created
Info : Hardware thread awareness created
Info : RP2040 Flash Bank Command
adapter speed: 1000 kHz

Info : CMSIS-DAP: SWD  Supported
Info : CMSIS-DAP: FW Version = 2.0.0
Info : CMSIS-DAP: Serial# = E661640843675228
Info : CMSIS-DAP: Interface Initialised (SWD)
Info : SWCLK/TCK = 0 SWDIO/TMS = 0 TDI = 0 TDO = 0 nTRST = 0 nRESET = 0
Info : CMSIS-DAP: Interface ready
Info : clock speed 1000 kHz
Info : SWD DPIDR 0x0bc12477
Info : SWD DLPIDR 0x00000001
Info : DAP init failed
in procedure 'program'
** OpenOCD init failed **
shutdown command invoked

*** [upload] Error 1```

For reference, here is my platformio.ini and main.cpp

```cpp
#include <Arduino.h>

void setup()
{
  pinMode(LED_BUILTIN, OUTPUT);
}

void loop()
{
  digitalWrite(LED_BUILTIN, HIGH);
  delay(500);
  digitalWrite(LED_BUILTIN, LOW);
  delay(500);
}
```

```toml
[env:pico]
platform = https://github.com/maxgerhardt/platform-raspberrypi.git
board = pico
framework = arduino
board_build.core = earlephilhower
debug_tool = cmsis-dap
upload_protocol = cmsis-dap
```

majbthrd, 2022-11-19:
Alas, I don't have a good answer for you.  I lack a Windows 11 platform to test with, and I'm loath to "upgrade" the one Windows 10 PC that I have which meets the Windows 11 system requirements.

In an earlier issue #18, a user was able to use Windows 11 after upgrading to the then latest pico-debug release (v10.04).

I would ask that you double-check the pico-debug release that you are using, and verify that (after loading the pico-debug .uf2) it appears in Device Manager without any warnings or errors.

ar1a, 2022-11-20:
> after upgrading to the then latest pico-debug release (v10.04).

just checked, and i've been using [10.05!](https://github.com/majbthrd/pico-debug/releases/tag/v10.05)

> verify that (after loading the pico-debug .uf2) it appears in Device Manager without any warnings or errors.

[image]
no errors! something to note is that pio is unable to upload with the above error yet if i use openocd manually:
```
PS C:\Users\aria\Downloads\xpack-openocd-0.11.0-5\bin> .\openocd.exe -f board/pico-debug.cfg
xPack OpenOCD x86_64 Open On-Chip Debugger 0.11.0+dev (2022-09-01-17:58)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
cortex_m reset_config sysresetreq

Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Info : CMSIS-DAP: SWD supported
Info : CMSIS-DAP: Atomic commands supported
Info : CMSIS-DAP: FW Version = 2.0.0
Info : CMSIS-DAP: Serial# = E661640843675228
Info : CMSIS-DAP: Interface Initialised (SWD)
Info : SWCLK/TCK = 0 SWDIO/TMS = 0 TDI = 0 TDO = 0 nTRST = 0 nRESET = 0
Info : CMSIS-DAP: Interface ready
Info : clock speed 4000 kHz
Info : SWD DPIDR 0x0bc12477
Info : [rp2040.cpu] Cortex-M0+ r0p1 processor detected
Info : [rp2040.cpu] target has 4 breakpoints, 2 watchpoints
Info : starting gdb server for rp2040.cpu on 3333
Info : Listening on port 3333 for gdb connections
``` it works just fine!

majbthrd, 2022-12-11:
Sorry, I just do not have the time to replicate the myriad of different environments of pico-debug users, and then investigate why that software doesn't play nice with pico-debug.

closed 2022-12-11 by majbthrd