majbthrd/pico-debug #27: Is there any way to use serial port in this daplink?

alwaysmy, 2022-12-07:
Hi,I wonder if there is some way to use uart in this virtual debug pod, so we can use the "printf()" function to print log in a serial monitor.

majbthrd, 2022-12-07:
No.

It feels like a fool's errand to try to modify RaspberryPi's printf implementation to support pico-debug and then try to beg and plead for them to incorporate the changes into their code.

Moreover, this approach would be the wrong way IMHO.  pico-debug gives the host PC direct access to all debugging information, so why would anyone want to emulate debugging using a serial UART?

closed 2022-12-07 by majbthrd

alwaysmy, 2022-12-07:
oh,ok,thanks,I just thought it's convenient when I do have a UART app like a console through serial or sth. else.